Grovkillen/ESP_Easy_Flasher #1: No COM ports avalible, COM port list empty

racer73, 2018-01-05:
Windows 10 1709 (OS Build 16299.125)

ESP Easy Flasher (0.00.013) executed from C:\ESP_Easy_Flasher

Programmer HW:
Silicon Labs CP210x at COM3

Application started ad Administrator. But no com port is shown in the list of avalible COM ports.

[image]

Grovkillen, 2018-01-06:
Thanks for the feedback. I'm in the makes of completely re-write the com scanner module and I have my hopes that this new method is more generic than the current one.

Grovkillen, 2018-01-06:
Please try the latest commit (0.00.014).

racer73, 2018-01-06:
Tested with 0.00.15, and it works. Great tool really handy.

closed 2018-01-06 by racer73

Grovkillen, 2018-01-06:
Good to know! Thanks! : Smiles

stafil, 2019-01-06:
Windows 10 1809

ESP Easy Flasher (ESP_Easy_Flasher-0.03.002) executed from d:\ESP_Easy_Flasher

Programmer HW:
Silicon Labs CP210x at COM3

Application started ad Administrator. But no com port is shown in the list of avalible COM ports.

Grovkillen, 2019-01-06:
And if you uncheck "only active ports"? Not in the list?

stafil, 2019-01-06:
check is disabled. nothing in list. only look searching.

Grovkillen, 2019-01-06:
Ok, I think I know what's wrong. I'll look into it tomorrow.

Grovkillen, 2019-01-08:
@stafil could you provide me with a registry dump for me to look at?

stafil, 2019-01-08:
######2019-01-08######
#######0.03.002#######
######FLASH INFO######
BIN file: Programming only...
COM port: **Please wait while scanning...**
Baud rate: 115200
######POST FLASH######
Unit name: 
Unit number: 
Admin password: 
WiFi SSID 1: 
WiFi password 1: 
WiFi SSID 2: 
WiFi password 2: 
IP: <DHCP>
######POST FLASH######
############################
[2019-01-08 16:36:24] JOB COMPLETED, over and out!

stafil, 2019-01-08:
still write: Please wait while scanning...

stafil, 2019-01-08:
recreate file COM_SCAN821.txt ....

stafil, 2019-01-08:
COM_SCAN3582.txt

stafil, 2019-01-08:
after 15 minutes com found. super. why so long? why search 10000 ports???

Grovkillen, 2019-01-08:
> DONE!
> (You may disconnect the unit if you want.)
> COM port is already in use. Cannot proceed.
> Adding settings using SERIAL commands.

I need to understand where in the registry your com port is located. It
seems that something was overlooked... I will add a stop after 200 ports or
something...

Sorry for the bug and thanks for the test :)

ohminy, 2020-01-14:
I have same issue in v0.04.007
Please wait while scanning...
recreate file... now COM_SCAN2939.txt..
What should I do? plz help me..

Grovkillen, 2020-01-14:
The problem is that the registry might have named the usb port something that my parser don't understand. You might want to test another usb port?

I will create a new flasher once the GUI has settled, that one will be universal for all operating systems and work a lot better. Sorry for the trouble....

rhopkins7428, 2020-01-19:
I was having the same exact problem with 0.04.007 my D1 mini was showing up as COM5 in my device manager, but the program said I had no active USB ports.  I tried moving the device to different connections on my PC, but it didn't work.  ***_*I was able to get it working by editing the COM port in device manager, changing it to COM2**_** and then rescanning with ESPEasyFlasher, now it was able to find it!

CompassTheSolution, 2020-03-14:
Well, my attempt to get started with ESP Easy ended rather abruptly as well.  Easy Flasher cannot locate my COM port which I've been using successfully with NodeMCU for years now.  I switched it from COM8 to COM1 and rescanned, but still not found.  If I uncheck "Only acitve ports", it does show me 3 ports that are all [offline], none of which is the one I need.

Suggestions?

CompassTheSolution, 2020-03-16:
Switched to a different USB/serial adapter and it was located.  Now I have to go back and figure out what the difference between the two is.  I thought they were identical...