Grovkillen/ESP_Easy_Flasher #3: Alpha  ?

Oxyandy, 2018-03-14:
Win10 x64, Regional Settings English AU, 
display resolution 1920 x 1080

 [image]

Grovkillen, 2018-03-15:
I know, it's the Win10 scaling that does that... sorry but I don't have a solution for it yet.

Oxyandy, 2018-03-17:
Any chance you could post a screen shot then so I can see all the text & then make use your program it look like a time saver, cheers

Grovkillen, 2018-03-17:
If you install the font, it'll look better

Oxyandy, 2018-03-17:
Sure, the font name will help 
thanks

Grovkillen, 2018-03-17:
It should be included with the exe? If it won't work it's the Agency FB
font.

Oxyandy, 2018-03-17:
Ok yes thanks, that worked - it is still slightly too tall for the screen, but everything is now visible - Thanks

closed 2018-05-01 by Oxyandy