Grovkillen/ESP_Easy_Flasher #5: Cannot open a PR

giig1967g, 2018-05-01:
Hi,
as I cannot open a PR on your project, I will add here the scp file.
[ESP_Easy_Flasher.scp.zip](https://github.com/Grovkillen/ESP_Easy_Flasher/files/1964508/ESP_Easy_Flasher.scp.zip)
If you like it just drop me a comment so I know it has been accepted :)

My suggestion is to be able to set the IP, unit, wifi credentials and password even if you don't upload the firmware. In other words to be able to set the POST FLASH activities even without loading a firmware.
To do that I have created a subroutine called START_POST_FLASH that is called if bin file is empty or after the firmware upload.

I couldn't compile the script to test, as I didn't understand the language.
please review my suggested proposal.

Grovkillen, 2018-07-16:
You can now do serial programing only as of 0.00.026

Grovkillen, 2018-07-16:
Thanks for the feedback and suggested code change! :)