Grovkillen/ESP_Easy_Flasher #7: Use standard Windows font

mrwee, 2018-09-05:
I really like this tool, but sometimes find the text a bit difficult to read. May I propose to use the standard Windows font, so the flasher adapts to the user's preferences?

Grovkillen, 2018-09-09:
I'm gonna release anb official tool later this autumn so no much time is give to this app for that reason.

Grovkillen, 2018-10-30:
Okay I might consider continuing on this app :) I'm considering the fact that I might do an overhaul of the GUI + letting the user tweak it a bit :)

localhost61, 2019-02-27:
I confirm that it's surprising ugly, but it looks better once the fonts or manually copied into the OS font folder as requested by the highlighted message in the application working area.
Here is what I first get on my Laptop (FHD resolution) with 0.4.7
[image]
And after restart when the fonts are installed
[image]

mrwee, 2019-02-27:
Yes, I've also found that installing the fonts, makes it better. However I'm not that crazy about the font itself, which was why I asked for the application to use the system font instead. That would be an great user option.

Grovkillen, 2019-02-27:
I will most likely turn it into a NodeJS app for multi OS support.

Grovkillen, 2021-01-18:
https://www.patreon.com/posts/46356158