Grovkillen/ESP_Easy_Flasher #8: COM port is already in use. Cannot proceed.

Domosapiens, 2018-09-15:
On WIN7 this is the log:

> DONE!
> (You may disconnect the unit if you want.)
> **COM port is already in use. Cannot proceed.**
> Adding settings using SERIAL commands.
> Flashing done (76.4s).
> Start flashing on COM18 (This might take a minute or two...)
> 
Any idea how to deal with this?

Grovkillen, 2018-09-18:
Sometimes you may need to do a manual reboot after the flash. Have you tried that?

Domosapiens, 2018-09-19:
Will try at the next occasion. Thanks

Domosapiens, 2018-10-29:
Manual reboot does not solve it -- COM port is already in use. Cannot proceed
Re-connect does not solve it -- COM port is already in use. Cannot proceed
Re-connect and then: Program only -- COM port is already in use. Cannot proceed
Re-connect and then: Open serial monitor -- COM port is already in use. Cannot proceed
Please advice (for WIN7)

Grovkillen, 2018-10-30:
That is really weird! I can only reproduce this if I have the serial monitor active or if the app has crashed before and is still keeping the port open... have you looked through the processes?

Grovkillen, 2018-11-01:
Is this still an issue?

Domosapiens, 2018-11-01:
@Grovkillen There is 1 ESP.Easy.Flasher.exe process running
Are there other processes I have to look for?

Flashing goes well, but post flash action fails.
So there is initial a correct connection between the app and the ESP for flash.

If I restart the app and go directly to Program only: the COM port is already in use.

Disconnect, connect the USB, removes and creates USB-Serial CH340 (COM21).
That one is recognized in the app.
When I push "Program only" I see shortly a new process dllhost.exe *32 COM surrogate
in addition to the dllhost.exe COM surrogate (without *32)  that is running all the time, with and without the USB plugged.

Does one of above ring any bell for you?

Grovkillen, 2018-11-01:
I will look into that. Could be some dependences not working correctly in
your environment.

Grovkillen, 2019-01-23:
Please update to 0.04.003 and report back here if this still is an issue.

closed 2019-01-23 by Grovkillen

bennybubble, 2019-03-26:
Using 0.04.007 and still an issue, using WIN10

marsic-84, 2019-04-05:
same problem(((

sascha777, 2019-04-24:
Just ran into the same problem.
Got it working since I opened up the Serial Monitor. After that it doesn't work anymore.
Tried reboot without success.
Any updates on this?

BagheeraHans, 2019-05-05:
Same issue here, used several board and flash tools.
On two different computers

zwinsch, 2019-06-09:
Same issue - with Win7 and (0.04.007)

Fgagnebien, 2020-01-23:
Ran into the same problem.
Using 0.04.007 and  WIN10.
Tried running as administrator also no difference, flashing goes well!
Still same issue,
COM port is already in use. Cannot proceed.

bbfendt, 2020-05-02:
Hello and thx for your great Software !
Same Problem for me here: "COM port is already in use. Cannot proceed"
Occurs after flashing with "post flash action" checked but also when just using "program only" or even when trying to open the Serial Monitor.
Only one instance running.
OS: Win10 x64
Version of Software: ß.ß4.007 and older versions too

nygellyndley, 2020-12-29:
I seem to be having the same issue.
Even just a straight "program" after a flash and esp reboot "COM port is already in use.  Cannot proceed."
Tried as admin

Grovkillen, 2020-12-30:
Yep I totally get the bug here, and I'm currently working on a better flasher tool which doesn't require you to download an exe and it will work on most types of host systems (may it be PC, tablet, phone or Chromebook).

// Jimmy Westberg
0706505709
٩(- ̮̮̃-̃)۶

Fr-Mike38, 2021-04-11:
Not a recent thread, but as it helped me today, I just share that all issues encountered when uploading firm on Sonoff where solved by :
- using a self-powered hub and short connections between USB-to-serial key and Sonoff PCB
- lauching ESP.Easy.Flasher.exe  in admin mode (W7)

With these two 'rules', upload runs correctly from begin to end.

If this can help somebody !