Grovkillen/ESP_Easy_Flasher #10: Error in:ESP.Easy.Flasher Cannot show dialog

Domosapiens, 2018-10-31:
@Grovkillen 
Just tried to execute the download 0.02
Run as administrator on WIN7
I see expanding to a number of sub-directories.
But with this error:
[image]

Important?

Grovkillen, 2018-10-31:
I know... I made a "small" error when updating the file. I will have it sorted this evening. You can hit abort and then restart and it'll not show up any more. I'm calling for a resource not there yet. If you use the file that is part of the nightly release in ESP Easy (today is the first one delivered with the flasher) I fixed it in that one but didn't have the time to update the repository.

Domosapiens, 2018-10-31:
Great. Thanks

closed 2018-10-31 by Domosapiens