Grovkillen/ESP_Easy_Flasher #11: Fixed IP seems not working

Domosapiens, 2018-11-03:
@Grovkillen 
Not sure if this an App problem or an ESPEasy problem

Using an old Vista PC. (WIN7 still not working)

Flash empty 4K goes well
Flash mega 20181101 goes well
Post flash action with no Admin password, Fixed IP and 2# SSID/Password
Post flash action seems to go well but ...
does not come out of the communication check wait loop at the end.

After termination and reboot:
the terminal shows ... not a valid wifi setting
terminal Settings command shows 0.0.0.0 as IP
terminal IP command accepts my input for a valid IP number.
terminal Save command
terminal Reboot command
After reboot
the terminal shows ... not a valid wifi setting
Settings command shows 0.0.0.0 as IP

**Is the terminal IP command not working ??? and therefore the App not working with fixed IP?**

Tried without the fixed IP
it does not come out of the communication check wait loop at the end.
After termination and reboot I need to use the terminal to discover the allocated IP,
to use that to change it to a fixed IP

Grovkillen, 2018-11-05:
I have found the issue... before the ESP would interpret the blank DNS and GW settings as a variant of the set IP... 

IP set to 192.168.1.**123** would render a DNS and GW setting as 192.168.1.**1**

Grovkillen, 2018-11-05:
We're working on it now... (adding commands for those settings within ESP Easy)

Domosapiens, 2018-11-05:
Great that you found something ..... but
If you set Esp IP to 192.168.1.123 (assuming GW is 192.168.1.1)
Then what is wrong with "would render a DNS and GW setting as 192.168.1.1"
That's what I have now. Is my current setting wrong?

[image]

I have a local time server on 192.168.1.7. (fixed)
For what do I need a DNS when using fixed IP's?
Is this a local DNS to discover and access the ESP's by name?
Since latest releases ... Note: Leave empty for DHCP ... an empty ESP DNS field is not an option anymore.

Grovkillen, 2018-11-05:
Your settings are correct but my flasher will not (currently) save it like
that. Only the IP field will be filled, the others will be blank and your
node will not be responsive. But we'll get this fixed very soon.

Grovkillen, 2018-11-05:
Fixed in 0.02.004

closed 2018-11-05 by Grovkillen