Grovkillen/ESP_Easy_Flasher #19: Rework GUI

pkejval, 2019-03-28:
Please, rework GUI completely. Use some Windows guidelines. Why is GUI using custom font? GUI is absolutely user unfriendly and font is ugly.

Grovkillen, 2019-03-28:
How much do you want to pay? ;)

I'm taking an job to cover my losses from trying to get this project to
generate money.

The GUI was built for my own use and got used by more and more people. I
agree with the interface needing to be redesigned but time is scarce... 4
year old and a 6 year old children at home. And out of 50k+ users only 100
are sponsors. So I need to get money elsewhere.

Sorry for snapping but we really need the community to step up funding wise
or else it'll probably not be developing as it could be.

Grovkillen, 2021-01-18:
https://www.patreon.com/posts/46356158