Grovkillen/ESP_Easy_Flasher #20: on Windows:Com ports >com9 not found

Muller59, 2019-09-29:
Greetings,

i recently added a new usb to ttl device to my collection. When plugged in, windows device manager tells me it is recognized and sees it as com10.
i can use it with putty and the likes of that.
The ESP-Easy_Flasher does not see it when attached to the computer.
Is there a limit on the number of com ports it looks for?
i have tried multiple versions of the flasher, including the one in the latest ESPEasy release.

giddyhup, 2020-03-03:
I'm in a similar situation. Windows sees the ports and Putty can communicate. The flasher claims there are no active ports. Has there been a solution?

krachlatte, 2020-05-07:
Hi, had the same issue, i used usbdeview and devicemanager to uninstall all com port greater com1 followed by a restarted to get the tool to detect the programmer correctly.

bets regards

Grovkillen, 2021-01-18:
https://www.patreon.com/posts/46356158