Grovkillen/ESP_Easy_Flasher #21: Load config.dat post flashing

subcamera, 2019-10-28:
Is it possible do load configurationfile "config.dat (or any other) post flashing automatically?

giig1967g, 2019-10-28:
that would be a really nice and useful feature!

Grovkillen, 2019-10-28:
Yeah I agree, the thing is that I'm working really hard with the new GUI (which will be a very much modular thing = takes some brain cells to get rolling) and after that I plan on making a Node.JS app instead of this hack. :)

Misiu, 2019-11-18:
@Grovkillen so the next `ESP_Easy_Flasher` will be based on Electron?
I really like Your tool, it helped a lot when I started using ESPEasy.
It would be awesome to have a new version of Your tool that would work on every system and would now have resolution issues (I run ESP_EASY_Flasher yesterday on low-res laptop and the flash button was not visible).
There are also some issues that aren't critical but would be good to have them fixed.
No rush on that. The tool works great and I'm really grateful for it, but I think that the community could help with an Electron-based app.

Grovkillen, 2019-11-18:
Yep I will make it an electron app OR a PWA app if that is simpler. The idea is that you could then use it by simply surf to our site.

Misiu, 2019-11-18:
@Grovkillen but how can a website (PWA) access USB Port (COM) or filesystem? Won't this require some software on PC? Like http://chilipeppr.com/? They require [`Serial Port JSON Server`](https://github.com/chilipeppr/serial-port-json-server).
I think that Electron app would be better.
There is one big advantage - you will be able to download it, install and run on a PC that doesn't have internet access. With PWA approach you won't be able to download the installer to the USB thumb drive and install it elsewhere. But I might be wrong on that :)

Grovkillen, 2019-11-18:
A PWA is possible to "download" using modern browsers. It would "only" require that you have the browser installed since it will tap into that to do stuff on your computer. I think that it's a standard, thus only a PWA compatible browser is needed. So no internet connection is needed once it's downloaded.

A big advantage is that it will seemlessly be upgraded if you have an internet connection. The local app code will be replaced with the latest online code.

Grovkillen, 2019-11-18:
ChiliPeppr uses a huge amount of external libraries including the (in my opinion) obsolete jQuery library. They don't use latest web technology... I wouldn't want to maintain such a beast ;)

Misiu, 2019-11-18:
@Grovkillen I agree that PWA can be installed, but You can't download the installer and install it on another PC. Take a look at https://github.com/chilipeppr/serial-port-json-server or https://github.com/arduino/arduino-create-agent this app must be installed locally to be able to communicate between browser and board.

Grovkillen, 2019-11-18:
But ChiliPeppr isn't a PWA.

Misiu, 2019-11-18:
> But ChiliPeppr isn't a PWA.

I agree. But how do You want to communicate between PWA and file system, USB and com? You must somehow upload ESPEasy to the board.

I've built a couple of PWA enable websites, I used PWA mostly to have an offline version of the website, but I never build complex web apps with PWA support. I don't think it's possible to do "native" like things with PWA, because it is still a website running in the browser.

Grovkillen, 2019-11-18:
I will look further into the matter once I get there. I will use the same framework for the app as I use for the new GUI so converting into either PWA or Electron is not a big deal.

Misiu, 2019-11-18:
@Grovkillen and whats the framework name? :)
I'm currently learning React, but I don't mind learning others to be able to help with Flasher.

Grovkillen, 2019-11-18:
It's called "gui easy" and is built by me. It includes the CSS and JS framework and will not have any external dependencies.

Grovkillen, 2019-11-18:
We cannot afford using external libraries since they are more generic and thus "too big". And I don't really know what they are doing. Many use old legacy stuff which we don't want to carry around.

Misiu, 2019-11-18:
@Grovkillen that's true, unfortunately most microcontrollers supported by ESP Easy have limited space.
Looking forward to the new GUI and to your framework. I hope I will be able to help with development in any way.

Grovkillen, 2019-11-18:
Sounds good! We need more eyes on the code :)